InfiniTimeOrg/InfiniTime #2111: Watch sometimes doesn't sleep after chime

Ghoelian, 2024-08-28:
### Verification

- [X] I searched for similar bug reports (including closed issues) and found none was relevant.

### What happened?

Occasionally, when the screen turns on for a chime, it doesn't go back to sleep.

### Reproduction steps

Turn on chime
I have it set to every 30 minutes, haven't tried with hour yet.

Wait.

### More details?

Sometimes, when I wake up, the watch is completely dead. Looking at the gadgetbridge battery history I can see the battery drained from (sometimes) almost full to completely dead in the span of a few hours. I can't confirm if this is also because the screen got stuck, but that's what I'm guessing has happened.

I have seen the screen get stuck on a few times myself as well, while the watch was laying flat on my desk. So the wake/sleep gestures shouldn't have any effect on it then.

### Version

v1.14.0, also main

### Companion app

Gadgetbridge

mark9064, 2024-08-28:
Thanks for the detailed report. This is caused and fixed by #2108 which hopefully should be landing soon

mark9064, 2024-11-17:
Should be fixed now on main (and for the upcoming 1.15). Please do re-open/create a new issue if you still encounter problems!

closed 2024-11-17 by mark9064

Ghoelian, 2024-11-30:
Been using the main branch for a little while now on my watch, and so far the issue hasn't resurfaced. Seems good now!