InfiniTimeOrg/InfiniTime #2112: How to get raw PPG data?

marcinkosztolowicz, 2024-08-28:
### Verification

- [X] I searched for similar issues (including closed issues) and found none was relevant.

### Introduce the issue

Hello,

I would like to ask if there's a way to retrieve raw PPG (Photoplethysmography) data from a Pinetime watch running the InfiniTime firmware. I am aware that it's possible to access such data using the wasp-os system, but I'm specifically interested in doing this with InfiniTime.

Is there any existing method, API, or workaround that allows for accessing the raw PPG data? If not, are there any plans to implement this feature in the future?

Thank you in advance for your help!

### Preferred solution

Myabe some applicatoin that saves data heartrate data to file system.

### Version

v1.14.0

mark9064, 2024-08-28:
The `Hrs3300::ReadHrs()` method returns the current PPG reading. It probably makes more sense to edit the HeartRateTask though as that takes care of periodic acquisition for you
You'd have to implement a way to save it to the filesystem or make it available over BLE. Both are viable options, but BLE only makes sense for realtime transmission to a nearby device

This would be better as a discussion (https://github.com/InfiniTimeOrg/InfiniTime/discussions) rather than an issue BTW

marcinkosztolowicz, 2024-08-28:
https://github.com/InfiniTimeOrg/InfiniTime/discussions/2113

closed 2024-08-28 by marcinkosztolowicz