InfiniTimeOrg/InfiniTime #2114: Rendering bug

corppneq, 2024-08-28:
### Verification

- [X] I searched for similar bug reports (including closed issues) and found none was relevant.

### What happened?

The Pinetime renders it's ui off-screen.

### What should happen instead?

The Pinetime renders it's ui on-screen.

### Reproduction steps

Look at the watch.

### More details?

The Pinetime seem to render it's ui off-screen.

Bug
[image]

I found a closed issue that contains a picture with the same bug:
_Originally posted by @inhji in https://github.com/InfiniTimeOrg/InfiniTime/issues/1411#issuecomment-1299015213_
But the other issue is closed and I'm running the latest release.

Version
[image]

### Version

v1.14.0

### Companion app

Bangle.js Gadgetbridge 0.80.0a Commit 2a1dc9fca

mark9064, 2024-08-28:
Are you using chimes? Almost certainly #1411 and fixed by #2019

That issue is closed as it's been fixed, but the fix is landing in 1.15 which isn't released yet

corppneq, 2024-09-01:
> Are you using chimes? 

Turn out chimes where indeed on. I must have enabled them by accident. I don't even want that function on.

Can you point me to the CI so that I can grab myself the latest dev build to testrun it? Or do I have to build locally?

closed 2024-09-01 by corppneq

mark9064, 2024-09-01:
The latest CI build as of now is [here](https://github.com/InfiniTimeOrg/InfiniTime/actions/runs/10511180923), but do note that is it's not a tested release