InfiniTimeOrg/InfiniTime #2116: When HR measurement is running, steps aren't recorded

KhazAkar, 2024-08-31:
### Verification

- [X] I searched for similar bug reports (including closed issues) and found none was relevant.

### What happened?

Step counter is not updated when HR measurement is running 

### What should happen instead?

Both should be working at the same time

### Reproduction steps

1. Flash build for commit 2625ed39e5a5d1f2ff2bfe8780d06e37c86e886a
2. Start HR measurement 
3. Walk, shake, whatever

### More details?

1. Disabling HR measurement 
2. Always On Display (didn't worked)

### Version

2625ed39e5a5d1f2ff2bfe8780d06e37c86e886a

### Companion app

Gadgetbridge Nightly (21aff699a)

mark9064, 2025-06-28:
Can you still reproduce this with 1.15?

KhazAkar, 2025-06-30:
I'm currently on a build from 14th of march, 12:12:04. Will flash newest successful build from main/master.

KhazAkar, 2025-06-30:
Tested on terminal watchface - works, only thing that doesn't (but AFAIK is known) is that after screen goes blank, HR measurement drops to 0 and needs time to recover.

mark9064, 2025-06-30:
Ah yeah OK, I think I know what you're talking about. There are three things here
1. Steps aren't always recorded instantly. The motion sensor counts steps itself, and the algorithm it uses means that it can take a few seconds for new steps to be reported. I imagine it does this to avoid reporting false-positive steps
2. HR measurement stops when the screen turns off. This means that the algorithm is reinitialised when the screen is turned back on again, and it takes ~6 seconds until a measurement can be made. The algorithm is also reset when too much ambient light is detected.
3. HR measurement is affected by motion. When you're walking, measurement will be impossible sometimes due to the signal quality being poor so either zero will be reported or the previous value will be maintained i.e. it will appear to freeze

Does that explain your observations?

KhazAkar, 2025-07-02:
Yes, I think it does quite well. This ambient light detection is built into HR sensor, if I'm not mistaken, right?

I was testing it while wearing PineTime on my left wrist, some walking, sometimes hand standed "still" to constantly check if sensors behave as I was thinking they do.

Is there a plan to allow HR measurement running periodically while screen is blank?