InfiniTimeOrg/InfiniTime #2117: When "Always On" is activated, after screen is inactive, measurements aren't updated

KhazAkar, 2024-08-31:
### Verification

- [X] I searched for similar bug reports (including closed issues) and found none was relevant.

### What happened?

Steps count doesn't update in Always On Display state

### What should happen instead?

It should update, just like time is updated

### Reproduction steps

1. Flash main
2. Enable "Always On" in display settings
3. Click side button
4. Walk/shake/whatever

### More details?

Only waking device up helps. Same thing applies when device is connected to GB

### Version

2625ed39e5a5d1f2ff2bfe8780d06e37c86e886a

### Companion app

Gadgetbridge Nightly (21aff699a)

mark9064, 2024-08-31:
Fixed by #2106

Thanks for the report

KhazAkar, 2024-08-31:
You're welcome, but for now that PR is in draft state :/

mark9064, 2024-08-31:
It will be merged eventually, don't worry. It is only a draft currently because it depends on another PR, and it doesn't make sense for me to rebase onto an architecture where it doesn't actually work for the point of review

KhazAkar, 2024-08-31:
Okay then, thanks for your heads up that my issue is being taken care of already. Great to have responsive contributors to such nice firmware on my freshly bought PineTime watch :)

mark9064, 2024-11-17:
#2106 is now merged so should be all good :)

closed 2024-11-17 by mark9064