InfiniTimeOrg/InfiniTime #2119: Weather format: Add 3rd option, "Scientific", which would use kelvin

MorsMortium, 2024-09-07:
### Verification

- [X] I searched for similar issues (including closed issues) and found none was relevant.

### Introduce the issue

There are currently 2 options, "Metric" and "Imperial", for Celsius and Fahrenheit. I think some people could like to display temperature in kelvin.

### Version

f8f8993

mark9064, 2024-11-17:
OK silly question maybe but does anyone actually miss this as a feature? It would add more code complexity and use more space, and I get the impression that it might not be worth the maintenance effort - I don't know anyone who uses anything other than F or C day to day. No normal weather site offers any other unit either

vkareh, 2024-11-18:
I don't think this is a particularly useful feature - I don't see folks walking around looking at their watch to see how many Kelvins it is outside.

"metric-vs-imperial" is very much customary everyday ways of measuring ambiance temperature. But "scientific" is not, since that is contextual to what they are doing. Plenty of scientists use Celsius as part of their work. Others may use Rankine or Kelvin based on their needs - they all have their uses and merits and there is no one "scientific" scale to use in every day life.

I certainly would not trust my PineTime watch with scientific measurements either :laughing:

L3br4nd, 2024-12-07:
Well, Kelvin is the SI unit for temperature. 
It is the reference for unit conversion because it resembles the absolute temperature and it is the temperature scale which provides the easiest way to calculate temperature radiation. 
Sure, °C and °F are more common, but from my understanding it is the most scientific temperature scale.

mark9064, 2025-01-10:
I agree, and for any calculations with temperature I'd use Kelvin for sure. But I don't want to look at my watch and go, "ooooh, 264K today, better grab my jacket", and I'm not convinced anyone else does either. It's like having for metres per second on a road satnav - sure