InfiniTimeOrg/InfiniTime #2122: "Always on" mode causes "Digital face" to display in 12-hour mode even if 24-hour mode is set

jthiem, 2024-09-17:
### Verification

- [X] I searched for similar bug reports (including closed issues) and found none was relevant.

### What happened?

Enabling "Always on" switches the Digital face to 12-h mode

### What should happen instead?

Digital face should continue to display 24-h time

### Reproduction steps

1. Select Digital face, and 24-h mode
2. Enable "Always on"
3. Look at watchface, it's in 12-mode
(24-h mode continues to be selected, and disabling always on reverts it to 24-h mode)

### Version

1.14.0 Short rev d468a964, compiled locally

mark9064, 2024-09-17:
Just as a sanity check, could you try deleting your settings file on the flash and then reboot and try the reproduction steps again?

You can delete it with anything capable of editing the file system e.g ITD, https://infinitimeexplorer.netlify.app/

I can't reproduce on my testing build but I haven't tried the exact commit you listed there yet - have you got any other changes on your branch?

jthiem, 2024-09-17:
Thanks for the tip, I'll check out deleting the settings. https://infinitimeexplorer.netlify.app/ does not show any files to delete though, I'll try the linux version next. I have two minor tweaks on my clone, the Digital Watchface shows the Week number ("Week 37" atm), and the timer defaults to 5 min rather than 0.

mark9064, 2024-09-17:
[image]

According to the README you should see something like this - is there no settings.dat visible?

jthiem, 2024-09-17:
Thanks for the tip, I deleted settings.dat using https://infinitimeexplorer.netlify.app/, but so far it didn't change anything. My clone has two minor tweaks, the Digital Watchface shows the week nummber ("Week 37" atm), and the timer app defaults to 5 minutes at startup.

mark9064, 2024-09-17:
Have you tried rebooting after deleting it by holding the side button for 7s? This will cause it to load default settings on the next boot

I think those modifications shouldn't cause a problem here

mark9064, 2024-09-17:
Sounds like a good test plan, if you still manage to reproduce on the main branch link the commit here and I'll try it too :)

jthiem, 2024-09-18:
After a clean build, applying my own small patches (https://github.com/jthiem/InfiniTime), I still could not replicate the bug. It must have been an artifact from an unclean build. Thus I'm closing this issue.

closed 2024-09-18 by jthiem