InfiniTimeOrg/InfiniTime #2144: How to update 1.11.0 to 1.14.0

cr9christi9n, 2024-10-23:
### Verification

- [X] I searched for similar issues (including closed issues) and found none was relevant.

### Introduce the issue

My Infinitime wath is under 1.11 version and I use on my mobile phone (Fairphone 4) the soft GadgetBridge 0.81.0.
I'd like to update to the version 1.14 but I didn't found the file to do it through GadgetBridge.
Anu help?
Many thanks in advance.

rambonette, 2024-10-23:
Hi @cr9christi9n !
Updating is pretty straight forward. In the [1.14.0](https://github.com/InfiniTimeOrg/InfiniTime/releases/tag/1.14.0) download the `pinetime-mcuboot-app-dfu-1.14.0.zip` on your phone.
Make sure you have Gadgetbrige updated with a version grater or equal to 0.78 and PineTime is connected to you phone and charged.
Open Gadgetbridge > select the Devices tab > click on the three dots on you PineTime tile > File installer > select the .zip file you just downloaded. Update process should complete in few seconds.

Mark this issue as closed if this was helpful.
Friendly reminder: you should use this Issues section to report problems, not to ask for help ;)
For that there's the [Discussions](https://github.com/InfiniTimeOrg/InfiniTime/discussions) tab.

Cheers 🍺

cr9christi9n, 2024-10-24:
Hi and manythanks @rambonette.
I undestand also for your reminder ;o)

closed 2024-10-24 by cr9christi9n