InfiniTimeOrg/InfiniTime #2156: Alarm time not in sync with device time

cyberneel, 2024-11-03:
### Verification

- [X] I searched for similar issues (including closed issues) and found none was relevant.

### Introduce the issue

Essentially, the alarms seem to be based on timers, which work by creating a timer for x amount of time to get us to the desired alarm time. This means it's not locked to the device time once set. So, if something happens and the watch is a few minutes different than my phone, for example, and I update the device time, the timers no longer reflect the proper alarm time.

### Preferred solution

Maybe some sort of callback when device time is updated so apps that rely on device time rather than timers can update their timers' periods to reflect the change.

### Version

main

mark9064, 2024-11-03:
Is this a theoretical musing or an actual bug you can reproduce? SystemTask reschedules the alarm after every time change

cyberneel, 2024-11-03:
I noticed it with my app I was making, which was based on the same code as the alarm. I see what you are talking about. I guess the solution would be to put the controller reschedule methods under the OnNewTime message, right?

mark9064, 2024-11-03:
Yep, that sounds right 👍

mark9064, 2024-11-03:
It's a bit annoying that it's this way round but the RTOS has no knowledge of wall-clock time and that probably is a good architecture in terms of consistency

mark9064, 2024-11-11:
Can this issue be closed now? We ended up concluding there is no bug right?

mark9064, 2024-11-17:
Closing, please do open another issue if there is more to discuss

closed 2024-11-17 by mark9064