InfiniTimeOrg/InfiniTime #2160: colours get inverted

lmamane, 2024-11-05:
### Verification

- [X] I searched for similar bug reports (including closed issues) and found none was relevant.

### What happened?

all colours were inverted on the watch

### What should happen instead?

colours should have stayed the same as before

### Reproduction steps

I don't have reproducible reproduction steps. The watch vibrated, I looked at it, and all colours were wrong. Deep blue became some kind of pink, the notification screen was dark-on-light instead of light-on-dark, ... It happened on my second day of use (first day of "real world" use, first day was "toying around").

### More details?

The discussions on IRC seemed to describe this as a new issue that appeared during the 1.15 development cycle, but I got it in pristine 1.14.1, as came preinstalled on the watch, I never flashed another firmware (yet). I just validated the 1.14.1 firmware in the on-watch menu.

### Version

v1.14.1

### Companion app

Gadgetbridge 0.81.0 comit 16018879b from F-Droid

mark9064, 2024-11-05:
Yes you're correct there is a separate recent issue. The issue you're talking about is #1984 and was fixed by #2046 which is landing in 1.15

FintasticMan, 2024-11-06:
I've found pretty consistent reproduction steps for the new issue:

1. Set the shake to wake calibration to the lowest.
2. Open the Dice app.
3. Lower to sleep.
4. Shake
5. Profit (?)

I haven't tested the patch you send in the chat yet, but I'll keep you updated when I try it.

lmamane, 2024-11-06:
> The issue you're talking about is #1984 and was fixed by #2046 which is landing in 1.15

Ah, sorry, not sure how I missed it in my search.