InfiniTimeOrg/InfiniTime #2175: Alarm hours cannot consistently be incremented positively

irisahivt, 2024-11-28:
### Verification

- [X] I searched for similar bug reports (including closed issues) and found none was relevant.

### What happened?

Attempted to increase the hour value of the alarm, pressing the appropriate button instead incremented minutes. Same behaviour with timer, though minutes and seconds where appropriate.

### What should happen instead?

The hour/minute value of the alarm/timer functions should increment as expected.

### Reproduction steps

Honestly, I don't know. The bug appears to crop up and vanish at random, irrelevant of restarting the device or reflashing the firmware.

### More details?

I've flashed the device with the firmware several times now, and while the bug may abate for a short time, it will return unpredictably. Likewise, restarting the device by holding the button until the bootloader flashes onscreen has yielded no results.

I don't yet know enough to attempt a firmware solution.

### Version

v1.14.0

### Companion app

Gadgetbridge v0.82.1

mark9064, 2024-11-28:
Does this happen both ways round, or is always clicking hours and getting minutes?

irisahivt, 2024-11-28:
It's always clicking hours and getting minutes. Or conversely, for the timer, clicking minutes and getting seconds.

mark9064, 2024-11-28:
Since the each spinner is a completely separate component with their own state internally, from an initial look over it seems unlikely that they're interacting with each other - my guess would be a touch panel hardware issue

Could still totally be a software issue though, I'll keep an eye out for if I ever see this issue on my device

mark9064, 2024-11-28:
I wouldn't close just yet - you can check if it's a hardware issue using the paint app. If you can get a dot to appear where the hours plus button is by clicking where the minutes plus button is, it's definitely hardware. If not, it could be hardware or software. If you can reproduce this issue frequently, open the paint app every time before you want to change the alarm or timer and you should get your answer reasonably quickly

mark9064, 2024-11-28:
Unfortunately that's almost certainly a hardware issue with the touchpanel :( If your unit is within the 30 day warranty I would contact support ASAP

irisahivt, 2024-11-28:
Regrettably I'm past that point, but thank you for your help! I'll order a new unit when I get the chance, and use this one for dev testing since I don't have the dev kit.

closed 2024-11-28 by irisahivt

mark9064, 2024-11-28:
Ah that's a shame. If you wanted a dev kit anyway though I guess it's not the end of the world. You can unseal a sealed watch with a hot air gun / careful hairdryer usage if you want to use the debugging pins as well, there are some guides floating around (don't remember where, I haven't tried this myself)