InfiniTimeOrg/InfiniTime #2177: Support Battery Level Status BLE Characteristic

liamcharger, 2024-12-03:
I would like the Battery Level Status characteristic to be implemented (see 3.29 in this [spec](https://btprodspecificationrefs.blob.core.windows.net/gatt-specification-supplement/GATT_Specification_Supplement.pdf#3.29.3))

I'm working on some features in InfiniLink that require the charging state to be known.

At the same time, this may allow InfiniTime to show itself in the iOS battery widget, like it is shown here:
[image]

Ralimbahere, 2024-12-19:
I would love to have that!

liamcharger, 2025-01-05:
As it turns out, the battery characteristic is enough for iOS to display the device in the widget, but the connection needs to be encrypted. (When using ANCS, it appeared.)

[image]

cyberneel, 2025-01-07:
@liamcharger Is it possible for the icon to be some watch?

liamcharger, 2025-01-07:
In theory, the icon should be a watch when the GAP Appearance value is set to Generic Watch (0xC0). The value is set [here](https://github.com/InfiniTimeOrg/InfiniTime/blob/main/src/components/ble/NimbleController.cpp#L109). According to the doc, the value is currently being set for a generic mouse, which is incorrect, but when I switched it to 0xC0, it didn't change.

mark9064, 2025-01-10:
According to https://www.bluetooth.com/wp-content/uploads/Files/Specification/Assigned_Numbers.html, 0xC2 is smartwatch which would be correct. Where did you see generic mouse? I could totally be misreading this BTW, I don't find the spec legible...

liamcharger, 2025-01-10:
> According to https://www.bluetooth.com/wp-content/uploads/Files/Specification/Assigned_Numbers.html, 0xC2 is smartwatch which would be correct (Section 2.6.3).

Interesting, you're right, although CxC0 would still be generic watch. Smartwatch wasn't listed where I was looking

> Where did you see generic mouse?

I'm looking for it, but I can't find the spec. It wasn't the same one you linked, though...

mark9064, 2025-01-10:
I think it's correct. I have a sports watch supporting BLE as well, I've just checked and it uses 0xC1 which NRF connect displays as "sports watch" - checks out. But interestingly 0xC2 is displayed as "unknown". So perhaps 0xC2 isn't well supported for some reason?

You tried C0 right? And that didn't work with apple either? I wouldn't be against changing it if C0 appears to be the way to do it

liamcharger, 2025-01-10:
Both C0 and C2 didn't work; I just checked what InfiniLink was receiving, and I got a 0 (undefined). So for some reason, iOS isn't getting an appearance value.

mark9064, 2025-01-10:
The joys of the black box apple BLE stack appear endless!

Do you have another device to check if that can read the value? Just to confirm it's there at all

liamcharger, 2025-01-11:
I actually do have an old Android device I can use...I'll get back to you on what I find.