InfiniTimeOrg/InfiniTime #2178: Stability issues on 1.15.0

AgentConDier, 2024-12-04:
### Verification

- [X] I searched for similar bug reports (including closed issues) and found none was relevant.

### What happened?

Frequent crashes/freezes after upgrading to the 1.15.0 release

### Reproduction steps

The stability issues are intermittent; there is no single way to cause a reboot 100% of the time.
Here is a walkthrough of two crashes, starting from a fresh boot with default settings:

1.  Enable raise/lower wrist wakeup
2. Connect to the watch using Gadgetbridge
3. Switch to PineTimeStyle watchface
   (Tested turning the display on and off using the button as well as motion detection multiple times between steps)
4. Send a notification to the watch, dismiss it and turn of the display
5. Attept to turn the display back on. No reaction, Watchdog triggers a reboot

--

1. Connect to the watch using Gadgetbridge
2. Turn the display off
3. Send a notification to the watch
4. No reaction from the watch. A few seconds later, the Watchdog triggers

### More details?

Notifications are not the only way to cause a crash. I have also observed reboots in the following situations after upgrading to 1.15:

- Disconnecting (using blueman) from the Laptop booted into a Mint Live ISO after upgrading using `itd`
- Trying to upload resources using `itd`
- Turning the display on or off (mostly using the raise/lower wrist motion, but sometimes also when pressing the button)
- Sending weather data using Gadgetbridge

I feel like enabling/using more features reduces the stability, especially bluetooth and wakeup motions.

Weather information provided to Gadgetbridge by de.kaffeemitkoffein.tinyweatherforecastgermany (FDroid)
Update installed using `pinetime-mcuboot-app-dfu-1.15.0.zip` from the GitHub release.

### Version

v1.15.0

### Companion app

Gadgetbridge Nightly 0.82.0-5a0f1e46d

zyphlar, 2024-12-06:
I'm getting crashes in InfiniSim on 1.15.0 that I wasn't getting before, usually when going to sleep. Unsure how to debug to get a better stack trace.

```
info:  [systemtask] Going to sleep
info:  [LCD] Low power mode
terminate called after throwing an instance of 'std::runtime_error'
  what():  Mutex released without being held
Aborted (core dumped)
```

NeroBurner, 2024-12-06:
@zyphlar Please open an issue at infinisim repo. Let's debug infinisim there

zyphlar, 2024-12-07:
Will do. Interestingly after clearing cmake cache I encounter nearly zero crashes, may have been a build fluke.

AgentConDier, 2024-12-07:
Tried again with a fresh copy of the firmware. Same checksum. Still crashes randomly when using it normally.

```
mint@mint:~/Downloads$ sha256sum *.zip
723a54064302ba85392dd4b0a11670c3203c8c4bc4e2b51568abd0fc5624adad infinitime-resources-1.15.0.zip
c03dc842dd6071b7375fbcf6ba8dc469f99979433b242fa21d88c016b980def2 pinetime-mcuboot-app-dfu-1.15.0.zip
mint@mint:~/Downloads$ itctl fw upg -r infinitime-resources-1.15.0.zip -a pinetime-mcuboot-app-dfu-1.15.0.zip
Uploading navigation1.bin 14983 / 14983 B |------------------------------------------------------------| 100.00% 3m35s
375600 / 375608 B |------------------------------------------------------------------------------------>| 100.00% 7m3s
Transferred 375608 B in 7m2.843585774s. Remember to validate the new firmware in the InfiniTime settings.
```

I switched back to stable Gadgetbridge `0.82.1`, as expected that didn't make a difference.

Even with bluetooth and wakeup disabled it can still crash after a while, usually when turning the screen on or off.
Disconnecting the charging cradle can also make the watch lock up and do a watchdog reset.
I left it on my desk for an hour and it was fine, but crashed right after I put it on and switched watchtace/enabled motion wakeup.

Starting to wonder if this is a hardware issue, sometimes the watch also bootloops a few times when trying to switch between versions (by holding the button during the boot loader screen) or crashes right after booting v1.15. Then again, 1.14.1 has been mostly stable, only reboots or freezes a few times per month.

mark9064, 2024-12-11:
Hmm yeah the lack of reports from anyone else is a bit strange since all pinetimes should be the same, it's not impossible that yours has a HW issue then :(

Do you know anyone else with a pinetime? Could you try their one for a bit and see if you can get that one to crash too?

AgentConDier, 2024-12-11:
Nope, don't know anyone else with a PineTime unfortunately.
I've ordered a second one from the pine64.com store, I'll try with that unit once it arrives in a month or two.

zyphlar, 2024-12-11:
I loaded the latest release onto my watch and followed the reproduction steps and don't notice an issue yet

dcoolArne, 2024-12-12:
I got one Watchdog reboot after exiting the shake calib. screen but i wasn't able to trigger it again. After i exitet the app the watch stop responding to inputs and then rebooted after a few sekunds. I don't really know what triggerd it except that i mabey quit the app twice, button + wish at the same time.

AgentConDier, 2025-01-03:
The new watch arrived. I've updated it from 1.14.1 to 1.15.0 using `itd` as well as paired and [replaced](https://gadgetbridge.org/internals/automations/intents/#change-gadget-mac-address) it in Gadgetbridge. So far it's been perfectly stable, but that doesn't mean much with 1h of uptime.

While messing around with the old watch some more (switched software using the blue pine cone method twice), it did [this](https://github.com/user-attachments/assets/87226a49-89fe-41aa-8940-e4fce9bce941) for maybe 5 minutes before it booted normally again. Before that, I'd gotten more than 18 days of uptime on the 1.14.1 version. If anyone wants to take a look at the watch and figure out why it dislikes 1.15 in particular, I'll gladly ship it to them. Otherwise I'm going to shelve it as broken and forget about it.
_The pine cone being cut off vertically is a video artifact_

mark9064, 2025-01-10:
How's it been after a week of the new one?

I don't have the hardware skills to try and postmortem your old one, but it would be really interesting to know why!

AgentConDier, 2025-01-10:
Agreed. Uptime is now 6d22h, still working perfectly fine :thumbsup:

AgentConDier, 2025-05-01:
I'm closing the issue, the new watch has been perfectly stable so far. (Apart from me accidentally pressing the button long enough to cause a reboot once)

closed 2025-05-01 by AgentConDier