InfiniTimeOrg/InfiniTime #2179: Firmware update with pinetime-mcuboot-app-dfu-1.15.0.zip and GadgetBridge fails

cr9christi9n, 2024-12-04:
### Verification

- [X] I searched for similar bug reports (including closed issues) and found none was relevant.

### What happened?

Firmware update "error" on my watch

### What should happen instead?

Firmware updated on my watch

### Reproduction steps

From GadgetBridge installed on a Fairphone 4 I select File installation and seleect the update (see title 1.15.0).
I see on  my watch screen "firmware update 0%" and few seconds later "error"

For info my actual firmware is 1.14.0 and boot version 1.0.0

### More details?

After having selected the downloaded firmware on gadget bridge, it's written ... "this firmware is for the hardware version (unknown)"

### Version

1.14.0

### Companion app

GadgetBridge 0.82.1 / commit a 99564d03

mark9064, 2024-12-04:
Could you try rebooting your watch by holding the side button for >7seconds and then retry? Does the error persist?

mark9064, 2024-12-04:
Hmm, it sounds like a companion app issue. Could you try removing the bluetooth bond in android settings and repairing through gadgetbridge? You might want to back up your gadgetbridge data in case you lose step count history. Alternatively if you can try another app to update the watch e.g itd

cr9christi9n, 2024-12-04:
I uninstalled Gadgetbridge completely, cleaned all the data and reinstalled.
Everything is now OK, the update to 1.15.0 is perfect.

cr9christi9n, 2024-12-04:
Many thanks to everyone and hourra for InfiniTime watch ;o)

closed 2024-12-04 by cr9christi9n