InfiniTimeOrg/InfiniTime #2185: Kelvin to Celsius degree conversion doesn't work in the the weather service

tokiwe, 2024-12-06:
### Verification

- [X] I searched for similar bug reports (including closed issues) and found none was relevant.

### What happened?

Kelvin to Celsius degree conversion doesn't work in the the weather service.

### Reproduction steps

I use Gadgetbridge. Configure it provide weather data for InfiniTime, go to Debug -> Show Cached Weather Data and check any tempature in Kelvin there, for example "Current Temp: 270 K". Now test set the weather format in InfiniTime to both metric and Imperial. In imperial InfiniTime correctly show 270 °K as 26°F, but in metric wrongly -2°C (270°F=-3.15°C).

### Version

v1.15.0

### Companion app

Gadgetbridge v0.82.1

tokiwe, 2024-12-06:
I highly suggest fix this bug fastly and release v1.15.1. This is so visible bug for majority of users.

mark9064, 2024-12-07:
Thanks for the report :)

Am I reading this correctly: the calculation is wrong but only by 1°C?

vkareh, 2024-12-07:
I can confirm that there's a weirdness in the calculation from K to C. Strange that F is correct since we literally calculate it from C. I'll take a look at the code for both InfiniTime and Gadgetbridge next week and see what's up.

L3br4nd, 2024-12-07:
I checked it with Infinitime 1.15 and Gadgetbridge 0.82.1 at current temperature 281 K and my watch shows 8°C,  which is correct (281 -273 =8). 
Maybe the defect only affects negative °C values?

FintasticMan, 2024-12-07:
Ah, I see the problem. We don't handle negative numbers correctly in the code for rounding the stored temperature for displaying. This is a pretty simple fix, so I'll make a PR in the next few days.

tokiwee, 2024-12-07:
> I can confirm that there's a weirdness in the calculation from K to C. Strange that F is correct since we literally calculate it from C. I'll take a look at the code for both InfiniTime and Gadgetbridge next week and see what's up.

I think it should calculate F from K because every

tokiwee, 2024-12-07:
> I can confirm that there's a weirdness in the calculation from K to C. Strange that F is correct since we literally calculate it from C. I'll take a look at the code for both InfiniTime and Gadgetbridge next week and see what's up.

I think, F should be calculated directly from K. Every conversion include rounding and if we firstly convert K to C, round it, C to F and then again round it, the result will be wrong for sure.

@FintasticMan