InfiniTimeOrg/InfiniTime #2186: Double tap to wake up doesn't register quick or non-in-rhythm taps

tokiwe, 2024-12-06:
### Verification

- [X] I searched for similar bug reports (including closed issues) and found none was relevant.

### What happened?

A double tap to wake up feature doesn't register quick taps. I've to do it slowly, in a very precise rhythm.

### Reproduction steps

Enable a double tap to wake up feature and test it. I would say 50% of attempts to wake up the screen will fail, until you do it in a very precise rhythm. The same applies to turning off the screen with a double tap.

Not intuitive behavior at all. I thought the feature was broken/buggy for a long time, until yesterday I realized it's just really pedantic about timing of taps.

### Version

v1.15.0

mark9064, 2024-12-06:
This is part of the touch panel firmware, so it's not something that can easily be tuned. Maybe make sure your finger is leaving the screen fully between the taps?

tokiwee, 2024-12-07:
> This is part of the touch panel firmware, so it's not something that can easily be tuned unfortunately. Maybe make sure your finger is leaving the screen fully between the taps?

Can we stop using that feature by the firmware and just register two regular taps as a double tap?