InfiniTimeOrg/InfiniTime #2187: Screen always freeze in the pong game

tokiwe, 2024-12-06:
### Verification

- [X] I searched for similar bug reports (including closed issues) and found none was relevant.

### What happened?

The pong game is buggy/not playable. It starts okay, but always before 10 points the screen freeze and you will lose.

### Reproduction steps

Play the pong game for a while and the screen always freeze in only a minute. Is this some kind of power saving feature in the screen that suspend it occasionally? A very strange behavior😅

### Version

v1.15.0

mark9064, 2024-12-06:
See if it's this: open the paint app and move your finger slowly. Does it stop tracking after 5 seconds?

This is the touch panel entering a power saving mode AFAIK - you're on the money there. Definitely annoying - I've been meaning to look into there's a way to tune it. Seems there might be from the datasheet

mark9064, 2024-12-07:
I've found a fix for this - just need to check it doesn't have any nasty side effects. Will try post it soon :)

tokiwee, 2024-12-07:
> See if it's this: open the paint app and move your finger slowly. Does it stop tracking after 5 seconds?

Yeah, exactly that.

> This is the touch panel entering a power saving mode AFAIK - you're on the money there. Definitely annoying - I've been meaning to look into there's a way to tune it. Seems there might be from the datasheet

So, what is the point of this "feature"?😅 I think, it should to go power save only after a couple of seconds without integraation with finger. Is this a common practice in touch screens? Touch screens of phones do this also?

> I've found a fix for this - just need to check it doesn't have any nasty side effects. Will try post it soon :)

Excellent! Without working fix we should drop the pong game. How should it be playable for now? I always lose very fast because of the screen freeze.