InfiniTimeOrg/InfiniTime #2188: Reorganize apps

tokiwe, 2024-12-06:
### Verification

- [X] I searched for similar issues (including closed issues) and found none was relevant.

### Introduce the issue

I think we should reorganize the apps. User need stop watch, wake alarm, timer, foot steps, heart rate, media control, navigation and weather in their every day life, but not games in other hand. So please put all the games to the end of the app menu. It really doesn't make sense that you need to scroll down 3th page for the weather app and games locate on 2nd page.

### Preferred solution

I think, the order should be: stopwatch, timer, alarm, foot steps, media control, weather, map, metronome, drawing, dice, 2048, pong

L3br4nd, 2024-12-07:
I also find the current order annoying, especially the weather on the third page.  

You forgot some Apps in your list. 

My suggestion would be: 

1. Alarm, Stopwatch, Timer (Time-related Apps) 
2. Heartrate, Steps, Weather (Watch-face related Apps) 
3. Media, Metronome, Navigation (Tools) 
5. 2048, Dice, Pong (Games) 
6. Drawing (Art)

Nitrowing, 2024-12-07:
I'd drop Games and Art altogether

tokiwee, 2024-12-07:
Art should be before games🤔

 1. Alarm, Stopwatch, Timer (Time-related Apps) 
 2. Heartrate, Steps, Weather (Watch-face related Apps)
 3. Media, Metronome, Navigation (Tools)
 4. Drawing (Art)
 5. 2048, Dice, Pong (Games)

tokiwee, 2024-12-07:
Would it be possible made a config file for this to the SPI NOR flash? Then users could rearrange them (generate .bin file with Python script and send to InfiniTime with https://infinitimeexplorer.netlify.app) however they want.

L3br4nd, 2024-12-08:
> Art should be before games🤔
> 
>     1. Alarm, Stopwatch, Timer (Time-related Apps)
> 
>     2. Heartrate, Steps, Weather (Watch-face related Apps)
> 
>     3. Media, Metronome, Navigation (Tools)
> 
>     4. Drawing (Art)
> 
>     5. 2048, Dice, Pong (Games)

In my personsl opinion the drawing app is pretty useless since it is not possible to save the drawing. Also I'd try to have as many apps as possible on the 2nd page. So I support @Nitrowing's idea to merge them in one category.