InfiniTimeOrg/InfiniTime #2189: Where the issues about bugs on the weather service, a double tap feature and the pong game and went?

tokiwee, 2024-12-08:
Where the issues about bugs on the weather service, a double tap feature and the pong game and disappeared??

mark9064, 2024-12-09:
I can assure you that no issues have been deleted or otherwise removed by the InfiniTime team. If you have not deleted them yourself, you will need to get in touch with GitHub support

vkareh, 2024-12-09:
yeah that's weird... I definitely have those threads in my email:
[image]

but trying to get there I get a 404:
[image]

JF002, 2024-12-10:
As @mark9064 said, we haven't removed any of those issues. Github might have an anti-spam feature that removed them for some reasons?
Anyway, I suggest you contact Github support about this.
Feel also free to create them (maybe just one to begin with) to see what happens.