InfiniTimeOrg/InfiniTime #2210: Bound but won't connect to gadget bridge

PeterKW, 2024-12-27:
I'm struggling to work out what's happening / find good enough logs or error messages. I have an Android 14 Fairphone 5 that I'm trying to pair through gadget bridge to my new PineTime. I have reset and reinstalled gadget bridge and repaired the watch many times. 

It says bound to device and then can't connect. 

I've also tried the reset firmware option on the PineTime? 

I'm a bit stuck on what I could try next? 

Thank you very much,
P

PeterKW, 2024-12-27:
[image]

[image]

[image]

[image]

125~2.png …]()

[image]

[image]

[image]

[image]

PeterKW, 2024-12-27:
[image]

minacode, 2024-12-29:
> Edit: I have also tried on an Android 13 device with Gadgetbridge

And did that work? (I guess not)

In the case of multiple not working phones, the issue is more likely your watch. 

Can you flash an older firmware version and see if it works?

Also maybe try to delete all files on the watch. You will lose your settings, though.