InfiniTimeOrg/InfiniTime #2228: SPI EasyDMA UB

mark9064, 2025-01-16:
I was looking into https://github.com/InfiniTimeOrg/InfiniTime/blob/main/src/components/ble/DfuService.cpp#L390 and it turns out that the EasyDMA engine cannot read from flash! - See https://docs.nordicsemi.com/bundle/ps_nrf52832/page/easydma.html

Further reading: https://github.com/nrf-rs/nrf-hal/issues/37

There is potential UB across the codebase here - any SPI writes using constant buffers may be stored in flash and therefore be invalid. In particular, all of the display writes may be invalid if the compiler is moving constant arrays to flash (and not copying them to the stack first). Oddly all of the display commands seem to be fine (they clearly work!), while the DFU magic clearly isn't.

I don't know the exact method the compiler uses to decide the storage location. Does anyone know it?
It's certainly affected by `constexpr` etc.
I might try to analyse the compiler output, but I don't have the tooling to load the binary set up, and it wouldn't be as good as understanding the actual allocation algorithm.

I've got a few ideas here:
1. Add a check that hard crashes if a flash pointer gets passed into Spi::Write to prevent this from happening in the future
2. Introduce some kind of blocking write. A lot of these writes are small, but we still don't want to store this data in static RAM. So instead we can load them onto the stack, and then block until the write is complete (won't be waiting long as they are small). We need to wait as data in the stack frame can be overwritten as soon as the function returns by future function's stack frame.
3. An alternative to (2) would be allocating write buffers on the heap and then the SPI controller freeing them when done. In practice InfiniTime pretty much never runs out of heap, and when it does it usually hangs pretty quickly anyway (LVGL dies).

It looks like the nrf-hal team implemented (1), but only support blocking writes currently.

I'm sure I've missed other good options too, so I'd love any input on this.

pipe01, 2025-01-16:
We could add a macro that forces a variable to be on RAM, like (not tested) `#define FORCE_RAM __attribute__((section(".data")))`, and also add a check on SPI writes as you mentioned like `if (ptr < 0x20000000 || ptr >= 0x20001000)` (or `< __data_start__` and `>= __StackTop` to be more portable)

mark9064, 2025-01-17:
Agree, keeping constants that are used a lot where blocking would hurt in RAM is a good idea. I think if it's defined as a global variable (but not constant) it's already guaranteed to be in the data section right? For other less used things like the LCD config arrays, it'd be best to avoid moving these constants to the data section as RAM is pretty precious. So that's why I think storing them on the stack or heap temporarily might work better there.

The `__data_start__ / __StackTop` boundaries sound like a great way to do the checking