InfiniTimeOrg/InfiniTime #2241: Infineat battery indicator missing in always on display

ScanuNicco, 2025-02-03:
### Verification

- [x] I searched for similar bug reports (including closed issues) and found none was relevant.

### What happened?

When the watch goes to sleep, the battery level indicator disappears on the infineat face

### What should happen instead?

The battery level indicator should remain visible.

### Reproduction steps

- Enable always-on display
- Switch to infineat watch fact
- Use watch as normal until the issue occurs
    - The issue only occurs intermittently, but happens frequently enough that I observed it while writing this
    - It seems to mostly occur when the watch goes to sleep on its own, rather than when I press the button

### More details?

Occasionally, I will look at the watch without waking it up, and the battery indicator is completely missing. When I wake up the watch, either by pressing the button or lifting my wrist, the symbol remains missing. Swiping right to reveal the quick actions menu and then returning to the watch face brings it back

[Video of the issue](https://github.com/user-attachments/files/18645971/pinetime_battery_indicator_bug.tar.gz)
(Github says "failed to upload" if I try and upload the uncompressed version, even though it is the same size)

### Version

v1.15.0

### Companion app

Gadgetbridge

SteveAmor, 2025-02-14:
I can reproduce 100% of the time if AOD is selected. Set display timeout to something low like 5 seconds (but works for 30 secs too)

1.Leave on Infineat watch face.
2.Place on charger.
3.Screen dims and then goes to low brightness for AOD and immediately, the pinecone disappears.
4.Press the button and the display wakes with the pinecone.
5. GOTO 3 😁 

Thinking this is linked to #2245

mark9064, 2025-02-14:
The always on display is 8 colours only, so colours will be rounded down to the nearest share of 1-bit colour. So probably the contrast of the pinecone isn't high enough to become a different colour to the background it's on?

LinuxinaBit, 2025-07-07:
For me (and it seems for @ScanuNicco as well), that is not the problem.
When the watch face returns to the full color range after being woken up, the pinecone is *still missing*.
Additionally, when the watch first goes to sleep, the pinecone *remains displayed* on the always on display for a few minutes, disappearing much later.

mark9064, 2025-07-11:
Thanks for the detailed description both, with this information it's because the SPI flash is slept during AOD so whenever the pinecone is re-rendered (battery level change) it fails to load.

I'm not sure on the best way to resolve this. Maybe just don't sleep the flash when in AOD?

LinuxinaBit, 2025-07-11:
That seems like it might be a bad idea for battery life?

Might a good compromise to somehow wait for the SPI flash to wake up and then re-render the display automatically after the watch is woken up. Doesn't fully fix the issue, but it makes it a tad less annoying.

Another idea is to detect when it goes missing and replace the pinecone into some kind of image thay doesn't rely on the flash being awake to work; maybe a literal grey rectangle.

mark9064, 2025-07-11:
The SPI flash uses very little power when on, about 15uA. The power consumption in AOD is >200x this, so fortunately it'll make no difference to AOD battery life