InfiniTimeOrg/InfiniTime #2245: Quick settings screen crashes when placed on charge

SteveAmor, 2025-02-11:
### Verification

- [x] I searched for similar bug reports (including closed issues) and found none was relevant.

### What happened?

Turning notifications off, placing on charger and pressing button resets watchdog

### What should happen instead?

Watch should not crash

### Reproduction steps

Display timeout is 5 secs.
Go through the below quite quickly.
Reproduced 100% of the time.

Swipe watchface to access quick settings
Tap notifications to turn off (green to red)
Place on charger
Small charge icon is displayed at top of display
Display switches off after 5 seconds
Press button to switch on display (I wanted to be sure it was charging)
Display doesn't switch on*
Remove from charger
Press button to switch on Display
Watchdog timeout occurs
See watchface for a fraction of a second and then white to green pine cone
Last reset = wtdg

*I am away from my charger now so can't test other scenarios.

### More details?

No shake to wake
Backlight low
24h
Infinieat
Display timeout 5s
Bluetooth connected
Battery level was <50% when I first noticed this and was reproduced multiple times at 100% charge.
Button presses were quick clicks. i.e. I didn't hold the button for 7 seconds to force a watchdog timer reset.

### Version

Standard V1.15 2105a7b6

SteveAmor, 2025-02-14:
Extensive testing complete and issue updated.

mark9064, 2025-02-14:
Can you reproduce this with 1.14 as well? I'm guessing it's a new issue

I've been doing some font work recently so Infineat is completely broken for me right now! I need to get round to recreating the entire external filesystem on my watch to test this, so I'll get round to this when I can

SteveAmor, 2025-02-14:
> Can you reproduce this with 1.14 as well? I'm guessing it's a new issue
> 
> I've been doing some font work recently so Infineat is completely broken for me right now! I need to get round to recreating the entire external filesystem on my watch to test this, so I'll get round to this when I can

I can confirm that this was introduced in #1980. However, I believe the problem is in the Infineat watch face code where it is trying to update the batteryPercentRemaining, which I propose is happeninging before the battery level small pinecone lvgl object is fully loaded/created, as it's having to draw the whole watchface given you were not previously on the Infineat watchface.

SteveAmor, 2025-02-14:
Further to my findings in #2241 if I select AOD, and replicate the above steps, I can see that the Infineat display renders correctly after the display "goes to sleep" and the battery indicator shows it charging. BUT, pressing the button doesn't wake the watch (it's still on the charger base at this time) and will eventually crash if you press the button a few times.

SteveAmor, 2025-02-14:
In addition, if notifications are enabled and you place on the charger when you are on the quick setting screen and wait for the display to go off, if you then send a notification it will not display until you lift the watch off the charger.

Also, if chimes are enabled, they do not go off until the watch is lifted off the charger (assuming they should have gone off when on the charger). So, they were blocked at some point.

Note, you need to have the Infineat watch face and place it on the charger, not on the watch face so it automatically moves to the watch face as per #1980 for this edge case where things appear to get blocked until you lift the watch off the charger.  All other watch faces function as expected on or off the charger - assuming #1980 triggered a return to the watch face.

SteveAmor, 2025-02-14:
As it looked like it could be a blocking issue, if I remove https://github.com/InfiniTimeOrg/InfiniTime/blob/993118a3bc5400e0b09a58fa1aa0bc8ae72634a2/src/displayapp/DisplayApp.cpp#L319 it fixes the problem(s) but the UX has the watch face transition glitch which is why it was added.

SteveAmor, 2025-02-15:
I have a workaround for this which I'll raise a PR for later today.

minacode, 2025-02-16:
Is this another data race on a (battery) controller by the system task and the display app?

SteveAmor, 2025-02-16:
> Is this another data race on a (battery) controller by the system task and the display app?

I don't believe it is a race condition as it is specifically related to this line https://github.com/InfiniTimeOrg/InfiniTime/blob/993118a3bc5400e0b09a58fa1aa0bc8ae72634a2/src/displayapp/screens/WatchFaceInfineat.cpp#L473 which continously draws the watchface so the code is blocked by https://github.com/InfiniTimeOrg/InfiniTime/blob/993118a3bc5400e0b09a58fa1aa0bc8ae72634a2/src/displayapp/DisplayApp.cpp#L319  No other watchfaces have an animation so are not blocked.

minacode, 2025-02-16:
The first line which draws the line depends on the battery percentage that is obtained from the battery controller. Can you verify that this value is valid?

SteveAmor, 2025-02-16:
If you comment out https://github.com/InfiniTimeOrg/InfiniTime/blob/993118a3bc5400e0b09a58fa1aa0bc8ae72634a2/src/displayapp/screens/WatchFaceInfineat.cpp#L473 it doesn't block so it's definitely the line being drawn.

minacode, 2025-02-16:
>  so it's definitely the line being drawn

Yes, but based on which values? Can it break if we pass it nonsense-data?

mark9064, 2025-02-16:
It's not the value being drawn that's the problem. It's the fact that it's being drawn every frame, and it takes longer than one frame to draw it. Therefore LVGL sits at 100% CPU and never finishes `lv_task_handler`, which results in the draw watchface spin loop locking up DisplayApp

SteveAmor, 2025-02-16:
If you place on charge while on the Infineat watchface and wait for the screen to switch off, the watch acts as normal.

SteveAmor, 2025-02-16:
> It's not the value being drawn that's the problem. It's the fact that it's being drawn every frame, and it takes longer than one frame to draw it. Therefore LVGL sits at 100% CPU and never finishes `lv_task_handler` with time to spare, which results in the draw watchface spin loop locking up DisplayApp

Is the whole watchface being rendered or just the small animation line?

The maths to calculate the position of the line will render in exactly the same place for several frames as the small pinecone is considerably less than 100px high. I guess we could check if that line position was previously drawn and not render it that frame?

mark9064, 2025-02-16:
I suspect the problem is that re-rendering requires loading the pinecone image from the external flash (which is very slow). How long does it take for the pinecone to complete a full cycle from bottom to top (with the screen just on normally while charging)?

SteveAmor, 2025-02-17:
> I suspect the problem is that re-rendering requires loading the pinecone image from the external flash (which is very slow). How long does it take for the pinecone to complete a full cycle from bottom to top (with the screen just on normally while charging)?

I discharged the watch to 25% (3.73v) and popped it on the charger. It took 3.35 secs (average over 3 times) for the animation to run from 25% to 100%.

SteveAmor, 2025-02-17:
> > It's not the value being drawn that's the problem. It's the fact that it's being drawn every frame, and it takes longer than one frame to draw it. Therefore LVGL sits at 100% CPU and never finishes `lv_task_handler` with time to spare, which results in the draw watchface spin loop locking up DisplayApp
> 
> The maths to calculate the position of the line will render in exactly the same place for several frames as the small pinecone is considerably less than 100px high. I'll investigate checking if that line position was previously drawn and not render it that frame refresh.

I wrapped an ```if (chargingBatteryPercent % 3 == 0)``` around https://github.com/InfiniTimeOrg/InfiniTime/blob/993118a3bc5400e0b09a58fa1aa0bc8ae72634a2/src/displayapp/screens/WatchFaceInfineat.cpp#L442 and the animation is still slick and it doesn't block. I've not seen the resultant CPU utilisation in Infiniemu though.

What I'm not happy with is the magic number 3. This is 100% / height of pinecone which I'll address with a comment.

I do like the fact the issue is addressed in the watchface code, rather than the display app.

Should I raise a new PR so both options can be evaluated?

mark9064, 2025-02-17:
So it looks like drawing the pinecone takes about 50ms. With the current refresh period at 20ticks = 19.5ms (I'm calling it 20ms), what happens with it drawn every 3rd frame is the following (when screen on as usual):
```
lv_task_handler()
    0ms draw starts (frame 1)
    50ms draw finishes (frame 1)
    returns 0
lv_task_handler()
    50ms draw starts (frame 2)
    51ms draw finishes (frame 2)
    returns 19
vTaskDelay(19)
lv_task_handler()
    70ms draw starts (frame 3)
    71ms draw finishes (frame 3)
    returns 19
vTaskDelay(19)
lv_task_handler()
    70ms draw starts (frame 4)
    120ms draw finishes (frame 4)
    returns 0
```
etc.

Theoretically drawing every other frame should be fine. But neither are great solutions: animations should never be tied to framerate.

Instead it'd be better to animate according to some wall-clock frequency e.g 5 times per second. Also, you probably want the time from the bottom of the animation to the top to be constant rather than the number of pixels per second to be constant, as otherwise at high percentages it'll blink between high values rapidly.

To achieve this, the watchface could store the previous time `Refresh()` has last executed, and also store the time since the animation started. Then if more than 200ms has elapsed since the last execution, you could set the value as the real elapsed time modulo the total animation time (e.g 4 seconds).

Not sure if that made sense, let me know :)

mark9064, 2025-02-17:
> Should I raise a new PR so both options can be evaluated?

Yeah definitely when we've settled on an implementation I think a new PR would be clearer review-wise

SteveAmor, 2025-02-18:
> So it looks like drawing the pinecone takes about 50ms. With the current refresh period at 20ticks = 19.5ms (I'm calling it 20ms), what happens with it drawn every 3rd frame is the following (when [...]

Absolutely makes sense and I was also thinking that being tied to framerate is bad. Thank you for providing the above. After many battery discharge processes, using the torch, I hard coded a battery percentage to view the animation and I think it looks better if it's based on pixels per second. Otherwise, when you get to the last few pixels, it will be very slow. (It's by sheer fluke that the animation currently looks to be a good speed). 
For a pixels per second approach, is it just as simple as setting the lv_task_create period to 200? And set ```chargingBatteryPercent += 3;``` due to the pinecone height?

minacode, 2025-02-18:
Nice job, finding that out.

Maybe you can write a (generic) animation class that handles the timing and drawing nicely and can be reused in other places.

There are also [animations in lvgl](https://docs.lvgl.io/7.11/overview/animation.html?highlight=animation). Maybe take a look at them as well.

SteveAmor, 2025-02-18:
Final solution I have created a PR for is based on lv_tick
Original CPU intensive animation from 0% to 100% was approx 5 secs.
Animation based on lv_tick rendering every 150 msecs results in 0% to 100% in approx 5 secs.

mark9064, 2025-02-26:
Fixed with #2256

closed 2025-02-26 by mark9064