InfiniTimeOrg/InfiniTime #2250: Burn-in after using always-on-display

ScanuNicco, 2025-02-16:
### Verification

- [x] I searched for similar issues (including closed issues) and found none was relevant.

### Introduce the issue

I got my pinetime a few weeks ago, and have been using it with always-on-display enabled since. I am beginning to notice a burn-in effect, where parts of the watch face (infineat) are visible on other screens, particularly notifications, but I can also make it out in the gray portions of the settings and quick-actions menus. I have also noticed it on the pong game and the pinecone loading screen at boot. It is currently pretty subtle, but I'm worried it could become worse over time. The effect is far less noticeable on a camera vs. looking at it directly, so I have included both an uneditied photo and one with the contrast adjusted to make the effect closer to what I'm seeing.

Notification Screen (Unedited):
[image]

Notification Screen (Contrast Adjusted):
[image]

Pong Game (Unedited):
[image]

### Preferred solution

I would like to see this investigated further, and if it's determined that this effect continues to get worse over time, possibly a warning in InfiniTime that pops up when you turn on always on display, saying that burn in may occur on some watches.

### Version

v1.15.0

mark9064, 2025-02-16:
I noticed this when implementing AOD. It's transient image retention and disappears with time (a few minutes) - try toggling the the flashlight app for a bit and any retention should be gone

mark9064, 2025-05-16:
Has it got any worse for you? Mine seems unchanged in intensity so far