InfiniTimeOrg/InfiniTime #2252: Infinitime sometimes crashes after alarm goes off

codeguylevi, 2025-02-16:
### Verification

- [x] I searched for similar bug reports (including closed issues) and found none was relevant.

### What happened?

Sometimes when my alarm goes off, I dismiss it and the watch screen freezes and won't respond to swipes or taps..

### What should happen instead?

Just to continue on working normally.

### Reproduction steps

It just happens sometimes when I dismiss it. I'm not sure what is wrong.

### More details?

You have to reboot it every single time it crashes.

### Version

1.15.0

mark9064, 2025-02-16:
I've got some good news, I believe this a duplicate of #2206 and it's already fixed for 1.16 in #2211 :)

I'll close this for now, but if you think you've found a different bug or you can reproduce crashes on a build with #2211 merged (such as the latest main), please do re-open

closed 2025-02-16 by mark9064

codeguylevi, 2025-03-01:
> I've got some good news, I believe this a duplicate of [#2206](https://github.com/InfiniTimeOrg/InfiniTime/issues/2206) and it's already fixed for 1.16 in [...]

When will you launch the next version?

mark9064, 2025-03-01:
There's no specific timeline, but from what I can it's probably more than a couple months out