InfiniTimeOrg/InfiniTime #2255: Time is off when disconnected

codeguylevi, 2025-02-17:
### Verification

- [x] I searched for similar issues (including closed issues) and found none was relevant.

### Introduce the issue

When infinitime is disconnected from the bluetooth application, the time gets off and is either too fast or too slow. It also won't update the time while it is connected.

### Preferred solution

If you can make sure a minute is exactly 60 seconds, maybe it wouldn't be off. Also, make it so it updates the time every time it is connected.

### Version

1.15.0

mark9064, 2025-02-17:
I have automatic time synchronisation in my companion app (Gadgetbridge) disabled and I get a few seconds per week of drift (which is what is expected for a crystal like the one in the PineTime).

Are you sure you're experiencing clock drift and not some other issue? What companion app are you using

Also, what do you mean by "it won't update the time while it is connected"?

mark9064, 2025-02-23:
To provide some numbers I've had 4 seconds of drift in the last 7 days

codeguylevi, 2025-02-23:
> I have automatic time synchronisation in my companion app (Gadgetbridge) disabled and I get a few seconds per week of drift (which is what is expected for a crystal like the one in the PineTime). [...]

@mark9064 I get drift. Also, what I mean that even though it is connected to the application, it won't update the time.

mark9064, 2025-02-24:
What companion app are you using?

codeguylevi, 2025-03-01:
@mark9064 Watchmate on the latest Ubuntu

codeguylevi, 2025-03-01:
I also get serious drift problems. Why can't a minute be exactly 60 seconds? Smartwatches should be more accurate.

mark9064, 2025-03-01:
Could you test this? Disable bluetooth on the watch, and note the exact time (with seconds) and also the wall clock time. Then 24 hours later, note the exact time along the wall clock time again. This information would allow the drift rate to be determined

codeguylevi, 2025-03-02:
I don't have a wall clock that shows seconds.

mark9064, 2025-03-02:
You could use an online clock like https://time.is/?

codeguylevi, 2025-03-14:
I don't have a phone; I am running the companion app on an old iMac that
was altered to run Ubuntu.

codeguylevi, 2025-03-14:
Which means I can't always sync it.