InfiniTimeOrg/InfiniTime #2274: Step counter should start counting even when WakeUpMode is set to DoubleTap

FOSSLover84, 2025-03-16:
### Verification

- [x] I searched for similar bug reports (including closed issues) and found none was relevant.

### What happened?

Do not set any automatic display wake up (i.e. only enable double-tapping). Then, only after first display wake-up of the day, the steps are counted starting from zero no matter how much you walked already with turned-off display..

### What should happen instead?

When having walked around with turned-off display, I expect to see the counted steps when turning on the display.

### Reproduction steps

Do not set any automatic display wake up (i.e. only enable double-tapping). Then walk around after the nightly counter reset, then double-tap to turn on the display.

### Version

1.15.0

### Companion app

N/A

SteveAmor, 2025-03-16:
Could be related to #787

FOSSLover84, 2025-04-05:
@SteveAmor Thank you for pointing me to #787. I have no idea why I did not find. Let's close this one then, it is exactly the same issue.

closed 2025-04-05 by FOSSLover84