InfiniTimeOrg/InfiniTime #2283: Watch reboots when shaking with a magnet near the bottom of the screen

Ghoelian, 2025-03-30:
### Verification

- [x] I searched for similar bug reports (including closed issues) and found none was relevant.

### What happened?

Watch reboots when shaking with a magnet near the bottom of the screen (isn't this what the title is for?)

### What should happen instead?

Probably nothing

### Reproduction steps

1. Turn on shake to wake
2. Hold a magnet near the bottom of the screen (when looking at it with the button on the left side)
3. Shake

### Version

v1.15.0

### Companion app

Gadgetbridge

Ghoelian, 2025-03-30:
[image]

tituscmd, 2025-07-10:
Can reproduce on custom 1.15.0

mark9064, 2025-07-11:
Can you still repro with #2329 ? If yes, try applying #2328 also

Ghoelian, 2025-07-11:
Can confirm the issue no longer happens with #2329

tituscmd, 2025-07-11:
> Can you still repro with [#2329](https://github.com/InfiniTimeOrg/InfiniTime/pull/2329) ? If so, try applying [#2328](https://github.com/InfiniTimeOrg/InfiniTime/pull/2328) also

I'm gonna try #2329 and see if it fixes it for me as well. Will update y'all in here 😄

mark9064, 2025-07-12:
Great :)

I don't know what part the magnet plays though, I guess it affects the accelerometer somehow?? Assuming that's the case (and that's a big assumption), my speculation is that it's caused by Wake/Sleep being triggered back to back continuously, causing the message queues to fill. With the Wake/Sleep pairs limited to being sent every 100ms the queues have time to clear.