InfiniTimeOrg/InfiniTime #2286: Support selection of applications or watchfaces at build time to cut flash usage

git4Chub, 2025-04-13:
### Verification

- [x] I searched for similar issues (including closed issues) and found none was relevant.

### Introduce the issue

I recently add a lot of new characters into font set which make internal flash space nearly full, and I want to cut some unlikely used applications or watchfaces to save flash space.

Although I tried compile parameter "-DENABLE_USERAPPS" to select the apps I need, the internal flash occupation do not decrease.

### Preferred solution

support some compile parameter to select applications or watchfaces at build time and cut flash usage

### Version

v1.15.0

mark9064, 2025-04-18:
The compile time parameter works fine for me, are you sure it's applying properly? If you build with say just stopwatch selected, does the build only have the stopwatch app when you run it?

Might be worth cleaning the CMake build cache as well

mark9064, 2025-06-28:
The build time selection feature definitely works for me and the flash decreases, so I'm going to close this for now. Feel free to re-open if you're still having issues

closed 2025-06-28 by mark9064