InfiniTimeOrg/InfiniTime #2290: Pressing alternating buttons really fast in stopwatch crashes the watch

Anonymous3-a, 2025-04-30:
### Verification

- [x] I searched for similar bug reports (including closed issues) and found none was relevant.

### What happened?

If you press the two stopwatch buttons repeatedly, the watch freezes and reboots

### What should happen instead?

It should just do what the buttons will do instead of crashing the watch.

### Reproduction steps

Press each button fast and repeatedly in the stopwatch app (use two fingers, one for each button). The watch will stop responding. A few seconds later, the screen will blink a little and the watch will reboot.

### More details?

N/A

### Version

1.14.1

### Companion app

N/A

mark9064, 2025-05-01:
Can you reproduce this on 1.15?

Anonymous3-a, 2025-05-07:
Yes. It would be very easy to reproduce, although it does throw the time off.

liamcharger, 2025-05-13:
I can confirm the issue is still present on 1.15

Anonymous3-a, 2025-06-09:
It's been a month, is there any progress? I'm guessing the bug is a input buffer overflow of some kind.