InfiniTimeOrg/InfiniTime #2291: Question: where is the Accelerometer app? or other trivial Accelerometer apps?

jurgensymynck, 2025-05-04:
Hi,

Here: https://docs.infinitime.io/en/latest/user-documentation/apps.html#accelerometer
and here: https://docs.infinitime.io/en/latest/user-documentation/index.html

... I find mention of an accelerometer app with this face: 
[image]

On my Pinetime watch (Infinitime  1.14.1, BMA425,) this app is missing.
It is not mentioned on most other pine64, Pinetime and Infinitime sites.
Also, browsing through this Infinitime repo and filtering on `Accelerometer`, `Accel` provides no mention on adding or removing this app.

I want to build an accelerometer based controller for my [hobby robot project](https://jurgensymynck.github.io/TiSCo_Docs/) using one or more Pinetime watches.

Can anyone point me in the direction of the source code for this elusive app, or maybe other (trivial) projects that use the accelerometers?

Thanks in advance!
gr. Jurgen

mark9064, 2025-05-04:
It's here https://github.com/InfiniTimeOrg/InfiniTime/blob/main/src/displayapp/screens/Motion.cpp

It got removed from the default build configuration anymore as it's a debug tool rather than a useful app, but you should be able to switch it on using the compile time app selection: https://github.com/InfiniTimeOrg/InfiniTime/blob/main/doc/code/Apps.md

PS use 1.15 as it has #1950 which will make your life easier if you want different sensor ranges :)

jurgensymynck, 2025-05-04:
thanks for the fast reply!
I just upgraded to stock 1.15, went effortless.
I will try building a custom firmware version and report back

gr. Jurgen

jurgensymynck, 2025-05-04:
It worked!

The tl'dr version is that I built the firmware using the docker development environment, after uncommenting following line in this file:

`.\src\displayapp\apps\CMakeLists.txt`

```cpp
if(DEFINED ENABLE_USERAPPS)
    set(USERAPP_TYPES ${ENABLE_USERAPPS} CACHE STRING "List of user apps to build into the firmware")
else ()
    ...
    set(DEFAULT_USER_APP_TYPES "${DEFAULT_USER_APP_TYPES}, Apps::Motion") # <- uncommented line
    ...
```

Thanks again, Now it's hacking time!
gr. Jurgen

closed 2025-05-04 by jurgensymynck