InfiniTimeOrg/InfiniTime #2293: Rebooting a watch resets the number of steps to 0

hank-milliken, 2025-05-12:
### Verification

- [x] I searched for similar bug reports (including closed issues) and found none was relevant.

### What happened?

Having some steps counted, then after a reboot of the watches the number resets

### What should happen instead?

Number of steps shall be persistent through reboot

### Reproduction steps

- Start device,
- count some steps
- reboot device by long pressing a button
- steps are 0

### Version

1.15.0

mark9064, 2025-05-12:
I've implemented this already but need to upstream it. Poke me after a while if I haven't yet :P

hank-milliken, 2025-06-04:
@mark9064 gently poking ;)