InfiniTimeOrg/InfiniTime #2295: Watch bond removed/overwritten

liamcharger, 2025-05-15:
When pairing a watch with an encrypted connection (you can use an ANCS build to create one), the watch often deletes/overwrites the bond it has with my iPhone. 

Let's say I pair my watch (without a previous bond) with all the persisted data removed from InfiniLink. When I connect, iOS will ask for the passkey to be inputted. I enter the passkey, and a bond is created. The next time the watch disconnects/reconnects, I get an error saying "There was an error connecting to the peripheral because the bond has been removed." To fix this, I have to delete the bond iOS saved, and pair again using the code. 

This behavior is pretty unpredictable--sometimes it happens, sometimes it doesn't. And it's not an issue with InfiniLink; all the pairing stuff is handled by iOS.

I'm not sure if it has something to do with the ANCS implementation, but I don't recall it happening when just encrypting the battery characteristic, although I could be wrong. 

Can anyone using iOS reproduce this?

mark9064, 2025-05-16:
Might just be my phone but it never seems to be bonded according to NRF connect (android), I can bond it manually but some time later it is unbonded again (I have not yet debugged the trigger of the unbonding). So I think there might be some weird behaviour happening

liamcharger, 2025-10-21:
Hopefully this is not another iOS issue...

If anyone else on iOS can reproduce this (like @tituscmd :) ), I'll look at reporting it to Apple. Maybe we can get some more reports on Android though

tituscmd, 2025-10-21:
Nope, just my one and only iPhone

liamcharger, 2025-10-21:
Got it. The reason I'm asking is because I occasionally pair it to my Mac during development, then pair to my phone for daily driving. 

The issue occurs when I try to connect via my phone (with an old bond saved on it), while the watch still has a bond saved from the laptop. Here, it's predictable and makes sense.

Anyway, I just wanted to make sure I wasn't having this problem solely  because I pair to different devices, and that this is a real issue someone else is experiencing