InfiniTimeOrg/InfiniTime #2334: Documentation for flashing with OpenOCD seems outdated

DanielVukelich, 2025-07-17:
### Verification

- [x] I searched for similar issues (including closed issues) and found none was relevant.

### Introduce the issue

[The documentation for how to flash with OpenOCD and STLink](https://github.com/InfiniTimeOrg/InfiniTime/blob/9afc23cba9bcf938d8c49d6e15e7662ee8e6385d/doc/openOCD.md) seems outdated.  It is many years old and cannot be followed as-written.

The ocd steps reference build artifacts that I can't find matching names for:
`<build_dir>/image-<version>.bin` and `<build_dir>/pinetime-graphics-<version>.bin`.  The bootloader file referenced by that document does exist in `build/output`

I am able to build infinitime, but these are the only `bin` files I find afterwards:

- ./bootloader/bootloader-5.0.4.bin
- ./bootloader/mynewt_nosemi_4.1.7.elf.bin
- ./build/CMakeFiles/3.22.1/CMakeDetermineCompilerABI_C.bin
- ./build/CMakeFiles/3.22.1/CMakeDetermineCompilerABI_CXX.bin
- ./build/src/resources/7segments_40.bin
- ./build/src/resources/lv_font_dots_40.bin
- ./build/src/resources/teko.bin
- ./build/src/resources/bebas.bin
- ./build/src/resources/7segments_115.bin
- ./build/src/resources/pine_small.bin
- ./build/src/resources/navigation1.bin
- ./build/src/resources/navigation0.bin
- ./build/src/pinetime-recovery-loader-1.15.0.bin
- ./build/src/pinetime-mcuboot-recovery-loader-1.15.0.bin
- ./build/src/pinetime-mcuboot-recovery-loader-image-1.15.0.bin
- ./build/src/pinetime-mcuboot-app-image-1.15.0.bin
- ./build/src/pinetime-recovery-1.15.0.bin
- ./build/src/pinetime-mcuboot-recovery-1.15.0.bin
- ./build/src/pinetime-mcuboot-recovery-image-1.15.0.bin
- ./build/src/pinetime-mcuboot-app-1.15.0.bin
- ./build/src/pinetime-app-1.15.0.bin
- ./build/output/bootloader.bin
- ./build/output/pinetime-mcuboot-app-image-1.15.0.bin
- ./build/output/pinetime-mcuboot-recovery-loader-image-1.15.0.bin
- ./build/output/src/pinetime-app-1.15.0.bin
- ./build/output/src/pinetime-mcuboot-app-1.15.0.bin
- ./build/output/src/pinetime-mcuboot-app-image-1.15.0.bin
- ./build/output/src/pinetime-mcuboot-recovery-1.15.0.bin
- ./build/output/src/pinetime-mcuboot-recovery-image-1.15.0.bin
- ./build/output/src/pinetime-mcuboot-recovery-loader-1.15.0.bin
- ./build/output/src/pinetime-mcuboot-recovery-loader-image-1.15.0.bin
- ./build/output/src/pinetime-recovery-1.15.0.bin
- ./build/output/src/pinetime-recovery-loader-1.15.0.bin

### Preferred solution

Update documentation to match the names of build artifacts

### Version

main branch build

epietrowicz, 2025-10-21:
@mark9064  Is there any update on which .bin files should be used when flashing the device via OpenOCD? Using `./build/output/bootloader.bin` and `./build/output/pinetime-mcuboot-app-image-1.15.0` seems to brick my device.

mark9064, 2025-10-21:
I think you want `pinetime-app-1.15.0.bin`? I've never flashed it directly though so I don't know any details of the process